ermtl/Open-Source-Ventilator #3: Configure the CI with GitHub Actions

ZakCodes, 2020-03-24:
It would be cool to have the CI configured for this project. I've configured the CI for an Arduino project on GitLab before by using [arduino-cli], which you can see [here](https://gitlab.com/chapeau-melon/controleur-serre/-/blob/master/.gitlab-ci.yml) (the project is french, sorry).

In the project I just linked, the CI includes the following thing:
* [ ] Compiling the project using [arduino-cli]
* [ ] Checking the code formatting against a code convention using [clang-format]
* [ ] Generating HTML documentation from the source code using [Doxygen] and publishing it to GitLab Pages (it could be published to GitHub Pages for this project)

It would also be cool to add a badge with the CI status in the README file like so many projects do like so:
```md
![Build Status](https://github.com/actions/hello-world/workflows/Greet%20Everyone/badge.svg)
<!-- 
This was taken from: 
https://help.github.com/en/actions/configuring-and-managing-workflows/configuring-a-workflow#adding-a-workflow-status-badge-to-your-repository 
-->
```

To do all this on GitHub, [GitHub Actions] is probably the best thing to use right now, since it is directly integrated into GitHub. However, there are other services available like [TravisCI](https://travis-ci.org/), [CircleCI](https://circleci.com/) and [Jenkins](https://jenkins.io/), but they require more configuration than [GitHub Actions].

To help with setting up the different stages of the workflow, the following [GitHub Actions] helpers would be good to use:
* [setup-arduino-cli](https://github.com/arduino/setup-arduino-cli): made by Arduino, configures [arduino-cli] to run in workflows
* [clang-format-action](https://github.com/cvra/clang-format-action): use clang-format to format C/C++ files using a coding standard
* [doxygen-action](https://github.com/mattnotmitt/doxygen-action): generate html docs using [Doxygen]

[GitHub Actions]: https://help.github.com/en/actions
[arduino-cli]: https://github.com/arduino/arduino-cli
[clang-format]: https://clang.llvm.org/docs/ClangFormat.html
[Doxygen]: http://www.doxygen.nl/

mattd3v, 2020-03-24:
I will look at doing this very soon!

mattd3v, 2020-04-09:
PR #31 sets up and compiles the project.

karx, 2020-04-17:
https://gitlab.com/open-source-ventilator/ventilator/OpenLung/-/issues/146 is a parallel issue to this over on GitLab.

karx, 2020-04-17:
To tackle the issue of `jm_wire` I created forks for the 2 libraries that needed to be updated.

* [BM280](https://github.com/karx/BME280)
* [LiquidCrystal_PCF8574](https://github.com/karx/LiquidCrystal_PCF8574)

And used git to clone these libraries instead of Arduino library manager. The [build](https://gitlab.com/kaaro/controller_ci_demo/pipelines/137341800) is passing now using this.

ladvine, 2020-04-26:
@karx When i use the original jm_Wire library, i am too facing issues.
I will make the CI build to use the libraries maintained in GitHub.

karx, 2020-04-27:
@ladvine awesome! 

So these builds will run on all PRs and add label?

ladvine, 2020-04-28:
> @ladvine awesome!
> 
> So these builds will run on all PRs and add label?

It will run on all PRs. But following things are still pending:-
1. Add label when build is successful.
2. Checking the code formatting against a code convention using clang-format
3. Generating HTML documentation from the source code using Doxygen and publishing it to GitLab Pages (it could be published to GitHub Pages for this project)

nimamoslemy, 2020-05-01:
Hi, we are a team that needs your feedback to complete our idea.
Please visit our pages and comment on it. 
[github](https://github.com/nimamoslemy/ventilator)
[gitlab](https://gitlab.com/nimamoslemy/avp-ventilator)