ermtl/Open-Source-Ventilator #9: A few more resources

Blimpyway, 2020-03-28:
Hi I'll add here links and docs I find useful, you decide whether you want them included in the project. 
A couple good tutorials about how to configure A4988 and DRV8825 stepper motor drivers: Most tricky one is current limit adjustment which is slightly different for the two drivers. 
- A4988:  https://www.makerguides.com/a4988-stepper-motor-driver-arduino-tutorial/
- DRV8825: https://www.makerguides.com/drv8825-stepper-motor-driver-arduino-tutorial/

Attached is a drv8825.pdf, which can replace A4988. You already have A4988 pdf 
[drv8825.pdf](https://github.com/ermtl/Open-Source-Ventilator/files/4397039/drv8825.pdf)
 
Program /circuitry should not be affected they can replace each other in 3d printer boards.

ermtl, 2020-03-30:
Thanks, there are several motor drivers that can be used with it. Drv8825 is interesting as it has higher voltage ability, however, latest calculations show we'll probably need a beefier motor and associated driver.

Something like this one:
https://www.aliexpress.com/item/32714985325.html

closed 2020-03-30 by ermtl

Blimpyway, 2020-04-01:
According to my [testsl](https://drive.google.com/drive/folders/1WHIIuMprFZx8wzUte5k5reL2CMoxuLD7?usp=sharing), a Nema17 with DRV8825 (and likely A4988 since it is tuned at under 1.2A) is fully capable to deliver pressure/volume/bpm needed.