ermtl/Open-Source-Ventilator #13: BME280 pressure 3d printed part

miguel5612, 2020-03-29:
Hi, we are trying to reply your respirator, but how to install bmp 180 into a tube?

ermtl, 2020-03-30:
It's a BMP280. As of now, the library implements BMP180 only for the secondary ambiant pressure sensor.
The BMP280 module is inside a syringe as you can see in the picture. The real one would be similar, but using a round cable passing in the middle of the elastic part of the syringe plunger so that it's in front of it, and the plunger makes it airtight. It's very important, else, humidity will always be 100% and wet electronics don't work too well for too long ...
The syringe I used contains a LUER connector and tube adapter. That's a standard connector for medical devices.
Most ambu-bags have a sensor outlet where you connect the tube. The sensor will stay in the machine, the tube is not a problem for the measurement as long as it's not kinked, same as commercial machines. We don't bring the sensor close to the patient to prevent exposure to electrical devices (that would require a special medical grade, expensive double insulation power supply) and it's also to prevent having long wires as I2C circuits can have all kind of troubles when connected over a distance.

[image]

closed 2020-03-30 by ermtl

miguel5612, 2020-03-30:
Excellent, Thanks :)

Blimpyway, 2020-04-01:
I wish it would use either BMP280 or BMP180 for both respiratory pressure and ambient one.  I mean in many places stocks of BMP280 might run out first and in other places BMP180 I'd hope to be able to go on with either one not being forced to use both. Not assuming in a global crisis global deliveries  work flawlessly so I got any part I order on digikey here tomorrow.

miguel5612, 2020-04-01:
Than you @Blimpyway 
Here is the code modified to allow BMP 180 👯‍♂ 
https://github.com/miguel5612/Arduino_Respirator/blob/master/OpenSourceVentilator/OpenSourceV
entilator.ino