ermtl/Open-Source-Ventilator #19: Please use cmH2O for units instead of pascals

gr5, 2020-03-31:
Good work so far by the way!

Pascals is metric and a wonderful unit.  However everyone is talking cmH2O.  Doctors, hardware designers, literature, other ventilators, everyone.  And I can trivially measure pressure by sticking a hose down into so many cm of water and making bubbles.

I know it's trivial to convert from one to the other but when you are debugging (like if you hit the P command for example) you want both internal variables and also displays to show everything in the same units.  There is less confusion, fewer bugs introduced, and debugging is slightly faster.

The exception might be the ambient (not ambiant by the way - it's misspelled somewhere) pressure.  This could be in any unit you prefer as it's the only thing we care about that is *absolute* pressure.  In this case pascals is fine, or mm of mercury or whatever.  Everything else will be relative pressure (relative to ambient or relative to each other).

Blimpyway, 2020-04-01:
Sensor libraries report absolute pressure in pascals. 

Medical equipment literature also refers to hPa (hecto-pascals = hundreds of pascals) which is equal to cmH20 . There are 1000 hPa and 1000cm H20 in an atmosphere.  

Anyway reporting a value within 1000 +/-50 cm range could also be confusing to someone expecting a few dozen cm.

ermtl, 2020-04-01:
I fully agree. I started with the sensor's native unit, but in the medical world, everyone uses cm H2O, the unit for pressure will be changed in the next version.
For 'ambiAnt' that's what happens when I mix up French and English ! Corrected, thanks.
atmospheric pressure will probably stay in Pa. Anyways, it's measured just to be subtracted from the actual measurement in cases where there is no relative pressure sensor (I'm waiting for some)

gr5, 2020-04-01:
You can probably abbreviate to "cm" if you want the unit in the variable name.  Although I see you don't seem to have that habit (which I like but I'm happy either way - sometimes known as hungarian notation or a variant - named after a hungarian software architect at microsoft).

@Blimpyway - hPa is okay but I prefer cmH2O even though they are only 2% different.  For absolute pressure, I agree Pa is a good choice for unit.

ermtl, 2020-04-02:
@gr5 The pressure unit have been changed as requested, see new V0.21. Closing the issue for now.

closed 2020-04-02 by ermtl