ermtl/Open-Source-Ventilator #26: No.sensor error 0

Bootpanic, 2020-04-03:
Getting intermittent "NO.SENSOR Error 0"

Stopping and starting seems to trigger error.

I have TM1638 for keys, display and bar graph.
BME280 sensor connected.

Bootpanic, 2020-04-03:
I'm using a Arduino Mega. Don't know if this could be related.

Bootpanic, 2020-04-04:
I have tried with an Arduino Uno. The problem remains the same.

ermtl, 2020-04-04:
First thing, there are 2 kinds of breakout boards for BME280. Some with 3.3V regulators that are designed for 3.3 to 5V, others without, only for 3.3V.
If you have the 3.3V version, connect it's supply to 3.3V, and the I2C pins will be ok (at least for a prototype) since they are open drain.
Then, you can activate #define ArduinoPlotter and within Arduino go to "Tools" / "Serial plotter", you should see the pressure trace (above the motor/cycle traces). If something goes wrong, you should see abnormal values. If the problem shows there, please share a screen capture.

You can also check the sensor with the examples included in the BME280 library.

Also make sure you use the jm_Wire.h library as explained here:
https://github.com/ermtl/Open-Source-Ventilator/blob/master/OpenSourceVentilator/README.md

Bootpanic, 2020-04-04:
First of all thanks for all the great work!

I'm using the Pimoroni BME280 breakout board whis is rated for a supply voltage from 2V to 6V.
The BME280 breakout board is connected the the Arduino 3.3V out.
[image]
https://shop.pimoroni.com/products/bme280-breakout.
Wire length is less than 15cm.
I have turned "#define debug_3      // print sensor datadata" on in the source and the sensordata in serial monitor looks OK when the "No.sensor error 0" is displayed.
I will try to connect the BME280 breakout board to the 5V supply to see if it makes any difference but i doubt it. I will report here as soon as i've tried.
Greetings,
Bootpanic

Bootpanic, 2020-04-04:
I just tried to connect the Pimoroni BME280 breakout to the Arduino 5V supply rail with no change. The error still occurs.  A friend of mine is having the same intermittent error. The error seems to be triggered when stopping/starting the ventilation.

ermtl, 2020-04-04:
This was not a sensor issue, but a mistake in the menu system. I made changes on the 2 lines 16 characters LCD display and and did not test them with the TM1638.
This is corrected in the new V0.23 version. I'll close the  issue for now, feel free to reopen  if needed

closed 2020-04-04 by ermtl