ermtl/Open-Source-Ventilator #29: Additional hardware options ( like the ones  in MIT E-Vent  )

GoDotMad, 2020-04-08:
I tried using this firmware but the required display and keys are not available here.
I know we could use the serial interface 
but it is not very intuitive. 
therefore can you add the hardware specified in MIT website? 
I tried doing that but I couldn't run the code using a test stepper while not connected to the machine 
it seams never executing the stepper.MoveTo(motorTarget) 
or it could be that I don't understand how it is ment to work.
one last thing why your need humidity and temperature readings ?
I'm planing to use mpx5050 pressure sensor 
and without patient mode

ermtl, 2020-04-08:
The display and the stepper control are independent, so this should not be the issue. 
Someone else (Billybrun) seems to be working on adding MIT style potentiometers. 
https://github.com/ermtl/Open-Source-Ventilator/issues/28
To start the ventilation process (motor), from the serial interface, you need to type the "1" command.

Here are all the serial commands:
                        "0 Stop assisted breathing\n"
                        "1 Start assisted breathing\n"
                        "CV sets Constant Volume mode\n"
                        "CP sets Constant Pressure mode\n"
                        "S set Speed (in BPM)\n"
                        "V set Volume (in ml)\n"
                        "P set Pressure (in cm H2O)\n"
                        "E set PEEP (in cm H2O)\n"
                        "A set Apnea alarm (in cycles)\n"
                        "X set eXpiration ratio (x inspiration)\n"
                        "Y set sYnc (in % of cycle)\n"
                        "M show Measurements from the sensor\n"
                        "T toggle Telemetry data on/off\n"
                        "W Write parameters to permanent memory\n"

Also, MPX5050 seems like a good choice.  If you add that and want it merged, please make sure the changes use conditional compilation so that the option can be activated or not.

Assisted breathing is disabled if the sync period is set to 0 (the respirator will trigger cycles at the programmed BPM without waiting for the patient to initiate a breathing cycle). Normally, both modes are important, but if you want to remove it, the  parameter ee.syncRatio should be set to 0

Humidity and temperature readings were designed as the sensor was originally supposed to be directly in the breathing path where they could be used for several purposes, including an alarm if the tube is not connected correctly to the patient, but since patient secretions would damage the sensor and bringing electrical devices in direct contact with the patient, this would have considerably complicated electrical safety rules and machine cost, so it should be moved in the control box and connected using a sensor tube (just like commercial ventilators). At this location, temperature and humidity are quite useless. The code is still there in case this is needed, but might become a conditional compilation.

GoDotMad, 2020-04-08:
thanks for the reply.
I figured the problem and although I thought it can't  happen to me anymore but here we are.
I connected the step to the direction of the stepper driver.

as for the potentiometers we too think that they are not suitable therefore we will replace them with a mechanical encoder and for the menus I suggest using https://github.com/neu-rah/ArduinoMenu library 

and once I feel I did a good job implementing that changes  to your already great code I will definitely make my first pull request.

closed 2020-04-08 by GoDotMad