ermtl/Open-Source-Ventilator #34: Please check this line of code

iotleaf, 2020-04-13:
@line 2026 (comparing values)
breathe==tick-breathLength;
did you mean to write (assigning values)
breathe=tick-breathLength;

ahogen, 2020-04-13:
FYI: https://help.github.com/en/github/managing-your-work-on-github/creating-a-permanent-link-to-a-code-snippet

https://github.com/ermtl/Open-Source-Ventilator/blob/fa3bfb8847b98581e2b42e0d2279bfaba44b49cb/OpenSourceVentilator/OpenSourceVentilator.ino#L2026

iotleaf, 2020-04-13:
Thanks ahogen

ermtl, 2020-04-13:
Damn ! it was there in plain sight, nice catch.
This caused a 1 breathing cycle delay before resuming breathing in power loss recovery and I searched for the cause for a while.

Fixed !
A major new version is on the way.

closed 2020-04-13 by ermtl