ermtl/Open-Source-Ventilator #35: Handle millis() rollover to zero

iotleaf, 2020-04-13:
I would check usage of millis() when comparing a previous_time & delay_period.

e.g. This will fail after the millis() timer rolls over to zero.
if ( millis() > previous_time + delay_period ) {}

Better to re-arrange
if ( millis() - previous_time >   delay_period ) {}

See
https://www.norwegiancreations.com/2018/10/arduino-tutorial-avoiding-the-overflow-issue-when-using-millis-and-micros/

grifarais, 2020-04-14:
Agreed. Probably better to use unsigned long type for all these timestamp and interval variables too for clarity.

ermtl, 2020-04-14:
The millis() rollover is addressed in a major new version that will be uploaded later today.
I have converted to unsigned long as suggested.

closed 2020-04-14 by ermtl