ermtl/Open-Source-Ventilator #36: compile fail

erez1977mit, 2020-04-16:
Hello,
I keep getting compilation errors in Arduino IDE (1.8.12) on two different computers (WIn10 & MAC).
The error is the same - exit status 1 - 'stepper' was not declared in this scope (see detailed log below)
Beside very few adaptation of the H.file to my configuration (add/remove //) -the file are originals.
I got this failure on version 0.3.
I downloaded all necessary libraries.
any help will be appericiated.

Error log:

/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino: In function 'void updateBargraph()':
OpenSourceVentilator:830:14: error: 'stepper' was not declared in this scope
    float vol=stepper.currentPosition()/motorVolumeRatio;
              ^~~~~~~
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino:830:14: note: suggested alternative: 'strupr'
    float vol=stepper.currentPosition()/motorVolumeRatio;
              ^~~~~~~
              strupr
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino: In function 'void Timer()':
OpenSourceVentilator:1595:51: error: 'stepper' was not declared in this scope
  if ((!stopMove) && (timerLvl<=interruptMaxNest)) stepper.run();
                                                   ^~~~~~~
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino:1595:51: note: suggested alternative: 'strupr'
  if ((!stopMove) && (timerLvl<=interruptMaxNest)) stepper.run();
                                                   ^~~~~~~
                                                   strupr
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino: In function 'void readSensors()':
OpenSourceVentilator:1793:11: error: 'stepper' was not declared in this scope
      if ((stepper.currentPosition()==0) && !ee.active)  // approximate athmospheric pressure by averaging when the bag is filled and the machine does not run.
           ^~~~~~~
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino:1793:11: note: suggested alternative: 'strupr'
      if ((stepper.currentPosition()==0) && !ee.active)  // approximate athmospheric pressure by averaging when the bag is filled and the machine does not run.
           ^~~~~~~
           strupr
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino: In function 'void setup()':
OpenSourceVentilator:1982:10: error: 'pin_Stepper_DIR' was not declared in this scope
  pinMode(pin_Stepper_DIR, OUTPUT);
          ^~~~~~~~~~~~~~~
OpenSourceVentilator:1984:10: error: 'pin_Stepper_Step' was not declared in this scope
  pinMode(pin_Stepper_Step, OUTPUT);
          ^~~~~~~~~~~~~~~~
OpenSourceVentilator:1987:10: error: 'pin_Stepper_Disable' was not declared in this scope
  pinMode(pin_Stepper_Disable, OUTPUT);
          ^~~~~~~~~~~~~~~~~~~
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino: In function 'void motorFastStop()':
OpenSourceVentilator:2168:2: error: 'stepper' was not declared in this scope
  stepper.setAcceleration(motorMaxAcceleration);
  ^~~~~~~
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino:2168:2: note: suggested alternative: 'strupr'
  stepper.setAcceleration(motorMaxAcceleration);
  ^~~~~~~
  strupr
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino: In function 'void basicTelemetry(boolean)':
OpenSourceVentilator:2195:45: error: 'stepper' was not declared in this scope
      Serial.print(arduinoPlotterMotorOffset+stepper.currentPosition());
                                             ^~~~~~~
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino:2195:45: note: suggested alternative: 'strupr'
      Serial.print(arduinoPlotterMotorOffset+stepper.currentPosition());
                                             ^~~~~~~
                                             strupr
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino: In function 'void breathingCycleStart()':
OpenSourceVentilator:2351:2: error: 'stepper' was not declared in this scope
  stepper.setMaxSpeed(i);
  ^~~~~~~
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino:2351:2: note: suggested alternative: 'strupr'
  stepper.setMaxSpeed(i);
  ^~~~~~~
  strupr
OpenSourceVentilator:2358:19: error: 'pin_Stepper_Disable' was not declared in this scope
      digitalWrite(pin_Stepper_Disable, LOW);  // activate
                   ^~~~~~~~~~~~~~~~~~~
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino: In function 'void breathingInspiration()':
OpenSourceVentilator:2388:12: error: 'stepper' was not declared in this scope
  long cpos=stepper.currentPosition ();
            ^~~~~~~
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino:2388:12: note: suggested alternative: 'strupr'
  long cpos=stepper.currentPosition ();
            ^~~~~~~
            strupr
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino: In function 'void breathOverpressureAlarm()':
OpenSourceVentilator:2440:8: error: 'stepper' was not declared in this scope
    if (stepper.currentPosition()< stepper.targetPosition ())
        ^~~~~~~
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino:2440:8: note: suggested alternative: 'strupr'
    if (stepper.currentPosition()< stepper.targetPosition ())
        ^~~~~~~
        strupr
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino: In function 'void breathingExpirationStart()':
OpenSourceVentilator:2463:11: error: 'stepper' was not declared in this scope
    long p=stepper.currentPosition()/motorVolumeRatio;
           ^~~~~~~
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino:2463:11: note: suggested alternative: 'strupr'
    long p=stepper.currentPosition()/motorVolumeRatio;
           ^~~~~~~
           strupr
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino: In function 'void loop()':
OpenSourceVentilator:2573:21: error: 'pin_Stepper_Disable' was not declared in this scope
        digitalWrite(pin_Stepper_Disable, HIGH);  // Deactivate
                     ^~~~~~~~~~~~~~~~~~~
OpenSourceVentilator:2578:56: error: 'stepper' was not declared in this scope
   if (checkMillis(breathe,breathLength+syncDelay) && ((stepper.currentPosition()>0) || stepper.isRunning())) breathe=millis()-(breathLength+syncDelay+50); // wait for the previous cycle to finish (safety)
                                                        ^~~~~~~
/Users/mac/Documents/Arduino/OpenSourceVentilator/OpenSourceVentilator.ino:2578:56: note: suggested alternative: 'strupr'
   if (checkMillis(breathe,breathLength+syncDelay) && ((stepper.currentPosition()>0) || stepper.isRunning())) breathe=millis()-(breathLength+syncDelay+50); // wait for the previous cycle to finish (safety)
                                                        ^~~~~~~
                                                        strupr
exit status 1
'stepper' was not declared in this scope

GoDotMad, 2020-04-16:
as it says in the comments in the code .
the new version uses modified accelstepper library 

thus you have to replace it with the file in libraries folder of the repository

ermtl, 2020-04-17:
That happens if you compile without #define stepDirMotor as you would then need to replace with another type of actuator, may that be a DC motor or pneumatic solenoids or others.
Note that you can leave it defined and test without connecting the motor if you don't have one

closed 2020-04-17 by ermtl