ermtl/Open-Source-Ventilator #37: is there any position feedback implemented?

GoDotMad, 2020-04-17:
I couldn't see in your demonstration nor inside the code a way for setting the starting position 
or limiting the motor movement other than the number of steps.
so correct me if I'm wrong. 
for example if the power goes off while the bag is compressed how does it go to the starting position in the next operation.

ermtl, 2020-04-17:
I'm currently working on a home 'switch' (but it will be optical, not a real switch as it needs a very long life).
The exact implementation is very dependant on your hardware.
the general idea is that the "switch" would be triggered every time the movement ends. 
If it does not get triggered on time, more pulses are issued, if it gets triggered too early, it stops, and that probably means the motor missed some steps. if the condition happens too often, an alarm should be triggered.

GoDotMad, 2020-04-17:
what about optical encoder as well?

ermtl, 2020-04-17:
it would be nice, but not very useful.
the compressed bag will act like a loaded spring. If the motor start losing steps, it means the pressure is way over what it should normally be or there is a mechanical problem and it will lose a lot of steps, trying to correct for it is almost hopeless and the best course of action is probably just to accept that the cycle is abnormal, detect it and try another cycle. If the problem persists, then alarm as there is not much else to do.
Servo motors/servo stepper controllers (except very expensive ones such as Yaskawa) would require an idle timeout or a system reset to restart after an error, so they are not appropriate.

closed 2020-04-18 by ermtl