ermtl/Open-Source-Ventilator #38: is this line supposed to be a comparison?

GoDotMad, 2020-04-22:
I think there an extra equal sign in this line of code

expectedPressurebyStep==compression/motorTarget;

similar to a previous issue

GoDotMad, 2020-04-22:
also  in the line " constrain( v,0.7,1 ); "
the returned value isn't used 
as well as " constrain( r,1.1,2.0 ) "

GoDotMad, 2020-04-22:
and one last thing
why isn't all timing variables of type unsigned long?
because I get a warning about comparison between signed and unsigned expressions

ermtl, 2020-04-22:
First issue, yes, thanks, that explain problems I had with control loop instability.
the constrain, the result is used in the condition on the line below.
For the unsigned values, you are right it's more consistent with unsigned everywhere, even if that won't make much of a difference since those that were not yet converted are for timings related to a cycle and they are typically only a few second long.

closed 2020-04-23 by GoDotMad