ermtl/Open-Source-Ventilator #39: How to determine the required speed and acceleration of the stepper motor?

GoDotMad, 2020-04-23:
your stepper seems like it is rotating multiple times each cycle and you mentioned the it is on microstepping. 
and therefore requires a high speed and acceleration 
so is the modification in accelstepper library that effective? 

my mechanical construction made it that the motor has to move a lot to compress the bag and thats why it skips some steps ( a lot ).

ermtl, 2020-04-24:
I'm modifying the accelstepper library now.
the released version (works with my hardware) is simply sending 2 pulses at a time, I'm working on a better one that also takes less time to execute and offers both higher step pulse speed and smoother operation (less noisy) at low speed.
The speed and acceleration are calculated in the breathingCycleStart function.
the amount of total movement can be changed with motorVolumeRatio and the non linearity can be compensated with the volume2pos function (and the online function calculator)

Many things could make your motor skip steps. I haven't seen your design, but most I have seen are very under powered. a NEMA17 motor will not work, a 3D printer driver won't work either. 12V won't work, 24V is marginal.
My current setup that not only moves, but keeps working under load (up to 60 cm H2O) at max speed (35 BPM) with max volume and 1/3 I/E ratio does a full stroke in 0.4 second and uses a 4NM motor, a LeadShine driver and a 36V 150W power supply.
The motor is this one:
https://www.automationtechnologiesinc.com/products-page/nema-23/kl23h2100-50-8b/
The controller is the DM882S (too big, but that's what I have):
https://www.aliexpress.com/item/32888516648.html
The DM556 (or maybe DM542S) would do:
https://www.aliexpress.com/item/32999575790.html
It must be a real Leadshine driver. cheap 10$ copies won't do.

My design moves about 3 turns. If yours moves a lot more than that, a servo motor will be needed as steppers lose power as their speed increases.

GoDotMad, 2020-04-24:
I know that the speed and acceleration for each cycle are calculated in starhbreathingsycle()

what I ment was the max speed and acceleration and volume to position ratio in motor parameters section of the header file.

as for the motor I'm using nema 23 but it can deliver 1.5 NM because its what available here

and we can't order parts from china or any other place in times before the pandemic let alone now.

thats why I had to make a mechanical kind of gearbox to increase the torque but that resulted in needing a lot of speed. 

so   should I just give up?

ermtl, 2020-04-24:
I have not tried to check the maximum value, all I know is as of now, I'm getting quite close to the maximum for processor use (that's why I am modifying the accelstepper library)
1.5NM is a bit weak. It can work for tests, but, depending on the gearing, there is a risk that it will be either too slow for max settings or not strong enough to work well under pressure.
If you need to go fast, using 48V might help.
Please note that my double pulse trick only works as long as the motor controller is at least set to half step. if it's already set to full step, doing 2 full steps at once won't work.
I don't know where you can get a stepper motor locally. Maybe find an industrial machine manufacturer or repair shop and ask them to sell you one or where they get their steppers/controllers.

closed 2020-04-24 by GoDotMad